microsoft/windows-drivers-rs #5: Generic Fail-able (and Taggable) Allocators

jxy-s, 2023-09-22:
Development in the Windows Kernel requires allocations from different pools and the capability to tag allocations is an invaluable tool for debugging/triage.

Today this repository looks to recommend a global allocator and one that allocates exclusively from non-paged pool with a hard-coded pool tag.

https://github.com/microsoft/windows-drivers-rs/blob/cd1fd23f000c64ea244b5cbfc33d64acb1104b8b/crates/wdk-alloc/src/lib.rs#L35-L45

Rust crates for the Windows Kernel probably shouldn't provide a global kernel allocator. Else everyone using these crates may subtly end up allocating memory using identical tags. This will make debugging and triage a nightmare in the long term. Additionally I'll note, the non-paged pools are far more limited resource that the paged pools. I recognize that the _safer_ option when implementing a global allocator is to force it into non-paged memory, since not doing so has the potential to introduce other issues. However I the resource exhaustion is far more likely when forcing all allocations into non-paged pools.

For Rust to be a first-class citizen in the Windows Kernel. The language must support generic fail-able allocators and the libraries for the Windows Kernel should expose and support appropriate allocators that are capable of specifying pool types and tags.

wmmc88, 2023-09-25:
Users of these crates are free to consume them without the [`alloc` feature](https://github.com/microsoft/windows-drivers-rs/blob/cd1fd23f000c64ea244b5cbfc33d64acb1104b8b/crates/wdk/Cargo.toml#L30), and define their own allocators with their own tags and pool. 

Rust requires a global allocator to be defined in order to use the types in the [`alloc` module](https://doc.rust-lang.org/alloc/index.html) in the standard library(ex. [alloc::vec](https://doc.rust-lang.org/alloc/vec/)). But this doesn't force all allocations to use that allocator, the user is still free to allocate themselves via the rust binding to `ExAllocatePool2`.

Ideally, we would like to support `alloc` types with different allocator "settings"(ie. pool and tag) each time, but this is currently not something that is supported in `stable` rust. In `unstable` rust, there are currently two approaches to provide `alloc` types with custom allocators: the [`allocator-api`](https://rust-lang.github.io/rfcs/1398-kinds-of-allocators.html) proposal (which some libraries implement on `stable` via [allocator_api2](https://github.com/zakarumych/allocator-api2)), and a [Storage-backed allocator proposal](https://github.com/rust-lang/wg-allocators/issues/93).

jxy-s, 2023-09-25:
> Rust requires a global allocator to be defined in order to use the types in the [alloc module](https://doc.rust-lang.org/alloc/index.html) in the standard library(ex. [...]

I think this ties back to https://github.com/microsoft/windows-drivers-rs/issues/6 and https://github.com/microsoft/windows-drivers-rs/issues/6#issuecomment-1732809169. [alloc::vec](https://doc.rust-lang.org/alloc/vec/) will abort the "program" if it can't allocate. I think that's an unacceptable outcome for the kernel. If the kernel can't allocate memory for some request it shouldn't bugcheck.

> Users of these crates are free to consume them without the [alloc feature](https://github.com/microsoft/windows-drivers-rs/blob/cd1fd23f000c64ea244b5cbfc33d64acb1104b8b/crates/wdk/Cargo.toml#L30), and define their own allocators with their own tags and pool.

As was said elegantly in the comment on the other issue, "I think it would be beneficial to consider options with how to make this a bit safer for developers to avoid some mishaps that are likely going to be very easy to make (e.g. accidentally using the wrong memory allocator)."

jdoubleu, 2026-08-30:
It would be really helpful if I could customize the tag used by [wdk-alloc](https://github.com/microsoft/windows-drivers-rs/blob/main/crates/wdk-alloc/src/lib.rs#L50) as a start, to prevent the issue that every rust driver uses the same tag.

When requiring it as a const generic parameter, we could even force every consumer of this library to specify their own tag:
```rust
#[cfg(not(test))]
#[global_allocator]
static GLOBAL_ALLOCATOR: WdkAllocator<u32::from_ne_bytes(*b"MYDV")> = WdkAllocator::<u32::from_ne_bytes(*b"MYDV")>;
```