microsoft/windows-drivers-rs #10: error

kouzhudong, 2023-09-25:
```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.7.4
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files\Microsoft Visual Studio\2022\Enterprise>cd /d D:\code\rust\code\windows\windows-drivers-rs

D:\code\rust\code\windows\windows-drivers-rs>cargo build
   Compiling wdk-sys v0.1.0 (D:\code\rust\code\windows\windows-drivers-rs\crates\wdk-sys)
error: failed to run custom build command for `wdk-sys v0.1.0 (D:\code\rust\code\windows\windows-drivers-rs\crates\wdk-sys)`

Caused by:
  process didn't exit successfully: `D:\code\rust\code\windows\windows-drivers-rs\target\debug\build\wdk-sys-bd0a85a523038361\build-script-build` (exit code: 101)
  --- stderr
  thread 'main' panicked at 'WDKContentRoot should be able to be detected. Ensure that the WDK is installed, or that the environment setup scripts in the eWDK have been run.', crates\wdk-build\src\lib.rs:111:64
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

D:\code\rust\code\windows\windows-drivers-rs>
```

kouzhudong, 2023-09-25:
```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.7.4
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files\Microsoft Visual Studio\2022\Enterprise>cd /d D:\code\rust\code\windows\windows-drivers-rs

D:\code\rust\code\windows\windows-drivers-rs>cargo make
[cargo-make] INFO - cargo make 0.37.1
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Running Task: workspace
D:\code\rust\code\windows\windows-drivers-rs
[cargo-make][1] INFO - Calling cargo metadata to extract project info
[cargo-make][1] INFO - Cargo metadata done
[cargo-make][1] INFO - Project: sample-kmdf-driver
[cargo-make][1] INFO - Build File: Makefile.toml
[cargo-make][1] INFO - Task: default
[cargo-make][1] INFO - Profile: development
[cargo-make][1] INFO - Skipping Task: legacy-migration
[cargo-make][1] INFO - Execute Command: "cargo" "build" "--profile" "dev"
   Compiling wdk-sys v0.1.0 (D:\code\rust\code\windows\windows-drivers-rs\crates\wdk-sys)
error: failed to run custom build command for `wdk-sys v0.1.0 (D:\code\rust\code\windows\windows-drivers-rs\crates\wdk-sys)`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `D:\code\rust\code\windows\windows-drivers-rs\target\debug\build\wdk-sys-86a04ece13d95311\build-script-build` (exit code: 101)
  --- stderr
  thread 'main' panicked at 'WDKContentRoot should be able to be detected. Ensure that the WDK is installed, or that the environment setup scripts in the eWDK have been run.', crates\wdk-build\src\lib.rs:111:64
  stack backtrace:
     0:     0x7ff72798cfac - std::sys_common::backtrace::_print::impl$0::fmt
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\sys_common\backtrace.rs:44
     1:     0x7ff7279af44b - core::fmt::rt::Argument::fmt
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\core\src\fmt\rt.rs:138
     2:     0x7ff7279af44b - core::fmt::write
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\core\src\fmt\mod.rs:1094
     3:     0x7ff727987b3f - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\io\mod.rs:1714
     4:     0x7ff72798cd5b - std::sys_common::backtrace::_print
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\sys_common\backtrace.rs:47
     5:     0x7ff72798cd5b - std::sys_common::backtrace::print
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\sys_common\backtrace.rs:34
     6:     0x7ff72798f82a - std::panicking::default_hook::closure$1
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\panicking.rs:269
     7:     0x7ff72798f47f - std::panicking::default_hook
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\panicking.rs:288
     8:     0x7ff72798fede - std::panicking::rust_panic_with_hook
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\panicking.rs:705
     9:     0x7ff72798fdcd - std::panicking::begin_panic_handler::closure$0
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\panicking.rs:597
    10:     0x7ff72798dbf9 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\sys_common\backtrace.rs:151
    11:     0x7ff72798fad0 - std::panicking::begin_panic_handler
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\panicking.rs:593
    12:     0x7ff7279b8b95 - core::panicking::panic_fmt
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\core\src\panicking.rs:67
    13:     0x7ff7279b8b52 - core::panicking::panic_display
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\core\src\panicking.rs:150
    14:     0x7ff7279b8b52 - core::panicking::panic_str
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\core\src\panicking.rs:134
    15:     0x7ff7279b8b52 - core::option::expect_failed
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\core\src\option.rs:1952
    16:     0x7ff72742f0e3 - core::option::Option<T>::expect::h228f7ddc97874d6f
    17:     0x7ff727429425 - <wdk_build::Config as core::default::Default>::default::h0a5c5c02d1374f4e
[... 4 lines cut ...]
    22:     0x7ff7279823c8 - std::rt::lang_start_internal::closure$2
                                 at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library\std\src\rt.rs:148
[... cut ...]
                                 at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    31:     0x7ffa0dcd7344 - BaseThreadInitThunk
    32:     0x7ffa0fc026b1 - RtlUserThreadStart
[cargo-make][1] ERROR - Error while executing command, exit code: 101
[cargo-make][1] WARN - Build Failed.
[cargo-make] ERROR - Error while running duckscript: Source: Unknown Line: 5 - Error while executing command, exit code: 1
[cargo-make] WARN - Build Failed.

D:\code\rust\code\windows\windows-drivers-rs>
```

pantsman0, 2023-09-25:
Just loading the Visual Studio environment isn't enough, you need to load the WDK too. The environment variable `WDKContentRoot` should exist and look something like `C:\Program Files (x86)\Windows Kits\10` or wherever it is is installed on your system.

kouzhudong, 2023-09-25:
```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.7.4
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files\Microsoft Visual Studio\2022\Enterprise>cd /d D:\code\rust\code\windows\Windows-rust-driver-samples

D:\code\rust\code\windows\Windows-rust-driver-samples>cargo make
[cargo-make] INFO - cargo make 0.37.1

D:\code\rust\code\windows\Windows-rust-driver-samples>cd "D:\code\rust\code\windows\Windows-rust-driver-samples"

D:\code\rust\code\windows\Windows-rust-driver-samples>pwsh.exe -Command "if ($env:CARGO_MAKE_CRATE_IS_WORKSPACE) { return };$cargoMakeURI = 'https://raw.githubusercontent.com/microsoft/windows-drivers-rs/main/rust-driver-makefile.toml';New-Item -ItemType Directory .cargo-make-loadscripts -Force;Invoke-RestMethod -Method GET -Uri $CargoMakeURI -OutFile $env:CARGO_MAKE_WORKSPACE_WORKING_DIRECTORY/.cargo-make-loadscripts/rust-driver-makefile.toml"
Could not resolve CoreCLR path. For more details, enable tracing by setting COREHOST_TRACE environment variable to 1
[cargo-make] ERROR - Error while executing command, exit code: -2147450745
[cargo-make] WARN - Build Failed.

D:\code\rust\code\windows\Windows-rust-driver-samples>
```

kouzhudong, 2023-09-25:
```
D:\code\rust\code\windows\windows-drivers-rs>cargo make
[cargo-make] INFO - cargo make 0.37.1
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Running Task: workspace
D:\code\rust\code\windows\windows-drivers-rs
[cargo-make][1] INFO - Calling cargo metadata to extract project info
[cargo-make][1] INFO - Cargo metadata done
[cargo-make][1] INFO - Project: sample-kmdf-driver
[cargo-make][1] INFO - Build File: Makefile.toml
[cargo-make][1] INFO - Task: default
[cargo-make][1] INFO - Profile: development
[cargo-make][1] INFO - Skipping Task: legacy-migration
[cargo-make][1] INFO - Execute Command: "cargo" "build" "--profile" "dev"
   Compiling proc-macro2 v1.0.66
   Compiling unicode-ident v1.0.11
   Compiling windows_x86_64_msvc v0.48.5
   Compiling winapi v0.3.9
   Compiling once_cell v1.18.0
   Compiling glob v0.3.1
   Compiling lazy_static v1.4.0
   Compiling libc v0.2.147
   Compiling rustversion v1.0.14
   Compiling prettyplease v0.2.15
   Compiling rustix v0.38.13
   Compiling serde v1.0.188
   Compiling thiserror v1.0.48
   Compiling windows-targets v0.48.5
   Compiling regex-syntax v0.7.5
   Compiling windows-sys v0.48.0
   Compiling clang-sys v1.6.1
   Compiling bitflags v2.4.0
   Compiling minimal-lexical v0.2.1
   Compiling memchr v2.6.3
   Compiling quote v1.0.33
   Compiling log v0.4.20
   Compiling syn v2.0.32
   Compiling nom v7.1.3
   Compiling bindgen v0.68.1
   Compiling either v1.9.0
   Compiling libloading v0.7.4
   Compiling serde_json v1.0.106
   Compiling regex-automata v0.3.8
   Compiling errno v0.3.3
   Compiling home v0.5.5
   Compiling which v4.4.2
   Compiling cexpr v0.6.0
   Compiling wdk-build v0.1.0 (D:\code\rust\code\windows\windows-drivers-rs\crates\wdk-build)
   Compiling windows-core v0.51.1
   Compiling regex v1.9.5
   Compiling itoa v1.0.9
   Compiling lazycell v1.3.0
   Compiling ryu v1.0.15
   Compiling rustc-hash v1.1.0
   Compiling peeking_take_while v0.1.2
   Compiling shlex v1.2.0
   Compiling windows v0.51.1
   Compiling cfg-if v1.0.0
   Compiling tracing-core v0.1.31
   Compiling overload v0.1.1
   Compiling tracing-log v0.1.3
   Compiling nu-ansi-term v0.46.0
   Compiling thread_local v1.1.7
   Compiling sharded-slab v0.1.4
   Compiling smallvec v1.11.0
[... 134 lines cut ...]
Testing driver package...

Signability test complete.

Errors:
None

Warnings:
None

Catalog generation complete.
D:\code\rust\code\windows\windows-drivers-rs\target\debug\package\sample_kmdf_driver.cat
[cargo-make][1] INFO - Running Task: infverif

D:\code\rust\code\windows\windows-drivers-rs\crates\sample-kmdf-driver>cd "D:\code\rust\code\windows\windows-drivers-rs\crates\sample-kmdf-driver"

D:\code\rust\code\windows\windows-drivers-rs\crates\sample-kmdf-driver>"\x64\infverif.exe" /v /w D:\code\rust\code\windows\windows-drivers-rs\target\debug\package\sample_kmdf_driver.inf /msft
系统找不到指定的路径。
[cargo-make][1] ERROR - Error while executing command, exit code: 1
[cargo-make][1] WARN - Build Failed.
[cargo-make] ERROR - Error while running duckscript: Source: Unknown Line: 5 - Error while executing command, exit code: 1
[cargo-make] WARN - Build Failed.

D:\code\rust\code\windows\windows-drivers-rs>
```

kouzhudong, 2023-09-25:
编译成功了：
```
D:\code\rust\code\windows\windows-drivers-rs>dir /a /s *.sys
 驱动器 D 中的卷没有标签。
 卷的序列号是 F2D4-19D5

 D:\code\rust\code\windows\windows-drivers-rs\target\debug 的目录

2023/09/25  22:19            69,632 sample_kmdf_driver.sys
               1 个文件         69,632 字节

 D:\code\rust\code\windows\windows-drivers-rs\target\debug\package 的目录

2023/09/25  22:19            69,632 sample_kmdf_driver.sys
               1 个文件         69,632 字节

     所列文件总数:
               2 个文件        139,264 字节
               0 个目录 469,106,626,560 可用字节
```

签名失败：
```
D:\code\rust\code\windows\windows-drivers-rs\crates\sample-kmdf-driver>"\x64\infverif.exe" /v /w D:\code\rust\code\windows\windows-drivers-rs\target\debug\package\sample_kmdf_driver.inf /msft
系统找不到指定的路径。
[cargo-make][1] ERROR - Error while executing command, exit code: 1
[cargo-make][1] WARN - Build Failed.
[cargo-make] ERROR - Error while running duckscript: Source: Unknown Line: 5 - Error while executing command, exit code: 1
[cargo-make] WARN - Build Failed.
```

closed 2023-09-25 by kouzhudong