microsoft/windows-drivers-rs #39: error[E0554]

kouzhudong, 2023-11-13:
```
error[E0554]: `#![feature]` may not be used on the stable release channel
 --> crates\wdk-macros\src\lib.rs:6:42
  |
6 | #![cfg_attr(feature = "nightly", feature(hint_must_use))]
  |                                          ^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0554`.
error: could not compile `wdk-macros` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
[cargo-make][1] ERROR - Error while executing command, exit code: 101
[cargo-make][1] WARN - Build Failed.
[cargo-make] ERROR - Error while running duckscript: Source: Unknown Line: 8 - Error while executing command, exit code: 1
[cargo-make] WARN - Build Failed.
```

wmmc88, 2023-11-13:
The `nightly` crate feature should only be used when using `nightly` toolchains. Can you provide the full log with the command you used to build? Are you enabling the nightly feature? What is the output of `rustup show`?

kouzhudong, 2023-11-13:
```
Microsoft Windows [版本 10.0.19045.3636]
(c) Microsoft Corporation。保留所有权利。

D:\git\github\forked\windows-drivers-rs>cargo clean

D:\git\github\forked\windows-drivers-rs>cargo make
[cargo-make] INFO - cargo make 0.37.2
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Running Task: workspace
D:\git\github\forked\windows-drivers-rs
[cargo-make][1] INFO - Calling cargo metadata to extract project info
[cargo-make][1] INFO - Cargo metadata done
[cargo-make][1] INFO - Project: sample-kmdf-driver
[cargo-make][1] INFO - Build File: Makefile.toml
[cargo-make][1] INFO - Task: default
[cargo-make][1] INFO - Profile: development
[cargo-make][1] INFO - Skipping Task: legacy-migration
[cargo-make][1] INFO - Execute Command: "cargo" "build" "--profile" "dev"
   Compiling proc-macro2 v1.0.66
   Compiling unicode-ident v1.0.11
   Compiling windows_x86_64_msvc v0.48.5
   Compiling winapi v0.3.9
   Compiling once_cell v1.18.0
   Compiling glob v0.3.1
   Compiling prettyplease v0.2.15
   Compiling rustversion v1.0.14
   Compiling libc v0.2.149
   Compiling lazy_static v1.4.0
   Compiling rustix v0.38.19
   Compiling serde v1.0.188
   Compiling memchr v2.6.3
   Compiling clang-sys v1.6.1
   Compiling regex-syntax v0.7.5
   Compiling windows-targets v0.48.5
   Compiling windows-sys v0.48.0
   Compiling quote v1.0.33
   Compiling minimal-lexical v0.2.1
   Compiling syn v2.0.32
   Compiling bitflags v2.4.0
   Compiling log v0.4.20
   Compiling thiserror v1.0.48
   Compiling nom v7.1.3
   Compiling libloading v0.7.4
   Compiling bindgen v0.68.1
   Compiling either v1.9.0
   Compiling serde_json v1.0.106
   Compiling regex-automata v0.3.8
   Compiling errno v0.3.3
   Compiling home v0.5.5
   Compiling wdk-build v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-build)
   Compiling which v4.4.2
   Compiling windows-core v0.51.1
   Compiling cexpr v0.6.0
   Compiling itoa v1.0.9
   Compiling lazycell v1.3.0
   Compiling rustc-hash v1.1.0
   Compiling regex v1.9.5
   Compiling ryu v1.0.15
   Compiling peeking_take_while v0.1.2
   Compiling shlex v1.2.0
   Compiling windows v0.51.1
   Compiling cfg-if v1.0.0
   Compiling tracing-core v0.1.31
   Compiling overload v0.1.1
[... 261 lines cut ...]
Warning: can't set `group_imports = StdExternalCrate`, unstable features are only available in nightly channel.
Warning: can't set `reorder_impl_items = true`, unstable features are only available in nightly channel.
Warning: can't set `version = Two`, unstable features are only available in nightly channel.
Warning: can't set `condense_wildcard_suffixes = true`, unstable features are only available in nightly channel.
Warning: can't set `unstable_features = true`, unstable features are only available in nightly channel.
[cargo-make][1] INFO - Skipping Task: format-toml-conditioned-flow
[cargo-make][1] INFO - Execute Command: "cargo" "build" "--all-features"
   Compiling thiserror v1.0.48
   Compiling wdk-macros v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-macros)
error[E0554]: `#![feature]` may not be used on the stable release channel
 --> crates\wdk-macros\src\lib.rs:6:42
  |
6 | #![cfg_attr(feature = "nightly", feature(hint_must_use))]
  |                                          ^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0554`.
error: could not compile `wdk-macros` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
[cargo-make][1] ERROR - Error while executing command, exit code: 101
[cargo-make][1] WARN - Build Failed.
[cargo-make] ERROR - Error while running duckscript: Source: Unknown Line: 8 - Error while executing command, exit code: 1
[cargo-make] WARN - Build Failed.

D:\git\github\forked\windows-drivers-rs>
```

kouzhudong, 2023-11-13:
```
D:\git\github\forked\windows-drivers-rs>rustup show
Default host: x86_64-pc-windows-msvc
rustup home:  C:\Users\Administrator\.rustup

installed toolchains
--------------------

stable-x86_64-pc-windows-msvc (default)
nightly-x86_64-pc-windows-msvc

active toolchain
----------------

stable-x86_64-pc-windows-msvc (default)
rustc 1.73.0 (cc66ad468 2023-10-03)

D:\git\github\forked\windows-drivers-rs>
```

wmmc88, 2023-11-14:
I see what's happening. you are running `cargo make` from the root. This by default enables cargo-make's [workspace support](https://github.com/sagiegurari/cargo-make#workspace-support), which tries to execute `cargo make` for each crate, using the crate's folder as the working directory. I am disabling this default behavior in #35 . See `default_to_workspace = false`.

As you can see in your log, sample-kmdf-driver actually finishes the packaging steps:
```
Successfully signed: D:\git\github\forked\windows-drivers-rs\target\debug\package\sample_kmdf_driver.cat

Number of files successfully Signed: 1
Number of warnings: 0
Number of errors: 0
```

To build and package sample-kmdf-driver, use `cargo make --cwd ./crates/sample-kmdf-driver` if you're in the repository's root. Alternatively, you can `cd ./crates/sample-kdmf-driver` first, then `cargo make`.

wmmc88, 2023-11-14:
https://github.com/microsoft/windows-drivers-rs/blob/main/CONTRIBUTING.md#build-and-test also outlines this

kouzhudong, 2023-11-14:
```
D:\git\github\forked\windows-drivers-rs>cargo build
   Compiling windows_x86_64_msvc v0.48.5
   Compiling proc-macro2 v1.0.66
   Compiling unicode-ident v1.0.11
   Compiling libc v0.2.149
   Compiling rustix v0.38.19
   Compiling glob v0.3.1
   Compiling prettyplease v0.2.15
   Compiling winapi v0.3.9
   Compiling once_cell v1.18.0
   Compiling serde v1.0.188
   Compiling windows-targets v0.48.5
   Compiling rustversion v1.0.14
   Compiling bindgen v0.68.1
   Compiling serde_json v1.0.106
   Compiling clang-sys v1.6.1
   Compiling windows-sys v0.48.0
   Compiling thiserror v1.0.48
   Compiling quote v1.0.33
   Compiling lazy_static v1.4.0
   Compiling log v0.4.20
   Compiling memchr v2.6.3
   Compiling syn v2.0.32
   Compiling bitflags v2.4.0
   Compiling minimal-lexical v0.2.1
   Compiling regex-syntax v0.7.5
   Compiling nom v7.1.3
   Compiling libloading v0.7.4
   Compiling either v1.9.0
   Compiling windows-core v0.51.1
   Compiling wdk-build v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-build)
   Compiling lazycell v1.3.0
   Compiling shlex v1.2.0
   Compiling errno v0.3.3
   Compiling home v0.5.5
   Compiling which v4.4.2
   Compiling ryu v1.0.15
   Compiling peeking_take_while v0.1.2
   Compiling itoa v1.0.9
   Compiling regex-automata v0.3.8
   Compiling cexpr v0.6.0
   Compiling rustc-hash v1.1.0
   Compiling windows v0.51.1
   Compiling cfg-if v1.0.0
   Compiling tracing-core v0.1.31
   Compiling regex v1.9.5
   Compiling overload v0.1.1
   Compiling tracing-log v0.1.3
   Compiling nu-ansi-term v0.46.0
   Compiling thread_local v1.1.7
   Compiling sharded-slab v0.1.4
   Compiling smallvec v1.11.0
   Compiling spin v0.5.2
   Compiling tracing-subscriber v0.3.17
   Compiling serde_derive v1.0.188
   Compiling thiserror-impl v1.0.48
   Compiling wdk-macros v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-macros)
   Compiling static_assertions v1.1.0
   Compiling wdk-panic v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-panic)
   Compiling wdk-sys v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-sys)
   Compiling wdk v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk)
   Compiling sample-kmdf-driver v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver)
   Compiling wdk-alloc v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-alloc)
    Finished dev [unoptimized + debuginfo] target(s) in 2m 15s

D:\git\github\forked\windows-drivers-rs>
```

kouzhudong, 2023-11-14:
```
D:\git\github\forked\windows-drivers-rs>cargo make --cwd ./crates/sample-kmdf-driver
[cargo-make] INFO - cargo make 0.37.2
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Project: sample-kmdf-driver
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Execute Command: "cargo" "build" "--profile" "dev"
   Compiling proc-macro2 v1.0.66
   Compiling windows_x86_64_msvc v0.48.5
   Compiling clang-sys v1.6.1
   Compiling prettyplease v0.2.15
   Compiling libc v0.2.149
   Compiling rustix v0.38.19
   Compiling serde v1.0.188
   Compiling serde_json v1.0.106
   Compiling bindgen v0.68.1
   Compiling wdk-build v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-build)
   Compiling windows-targets v0.48.5
   Compiling windows-sys v0.48.0
   Compiling windows-core v0.51.1
   Compiling quote v1.0.33
   Compiling syn v2.0.32
   Compiling windows v0.51.1
   Compiling errno v0.3.3
   Compiling home v0.5.5
   Compiling which v4.4.2
   Compiling serde_derive v1.0.188
   Compiling thiserror-impl v1.0.48
   Compiling wdk-macros v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-macros)
   Compiling thiserror v1.0.48
   Compiling wdk-sys v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-sys)
   Compiling wdk v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk)
   Compiling sample-kmdf-driver v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver)
   Compiling wdk-alloc v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-alloc)
    Finished dev [unoptimized + debuginfo] target(s) in 2m 01s
[cargo-make] INFO - Running Task: rename-dll-to-sys

D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver>cd "D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver"

D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver>echo "D:\git\github\forked\windows-drivers-rs\target\debug"
"D:\git\github\forked\windows-drivers-rs\target\debug"

D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver>cd "D:\git\github\forked\windows-drivers-rs\target\debug"

D:\git\github\forked\windows-drivers-rs\target\debug>mkdir package

D:\git\github\forked\windows-drivers-rs\target\debug>if exist package\sample_kmdf_driver.sys (del package\sample_kmdf_driver.sys )

D:\git\github\forked\windows-drivers-rs\target\debug>rename sample_kmdf_driver.dll sample_kmdf_driver.sys

D:\git\github\forked\windows-drivers-rs\target\debug>copy sample_kmdf_driver.sys package\sample_kmdf_driver.sys
已复制         1 个文件。
[cargo-make] INFO - Running Task: stampinf

D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver>cd "D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver"

D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver>copy "D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver\sample_kmdf_driver.inx" "D:\git\github\forked\windows-drivers-rs\target\debug\package\sample_kmdf_driver.inf"
已复制         1 个文件。

D:\git\github\forked\windows-drivers-rs\crates\sample-kmdf-driver>stampinf.exe -f "D:\git\github\forked\windows-drivers-rs\target\debug\package\sample_kmdf_driver.inf" -d * -a amd64 -c sample_kmdf_driver.cat -v * -k 1.33 -n
Stamping D:\git\github\forked\windows-drivers-rs\target\debug\package\sample_kmdf_driver.inf
Stamping [Version] section with DriverVer=11/14/2023,8.23.15.242
   stampinf: found Section-Line 6: [Version]
   stampinf: In Version section
   stampinf: found Section-Line 13: [DestinationDirs]
   stampinf: adding DriverVer= at end of Version section
[... 160 lines cut ...]
    Issued by: DriverCertificate
    Expires:   Sun Jan 01 07:59:59 2040
    SHA1 hash: 0482739779270EF1C5CB55642E7E71ABA85AB6B1

    Issued to: DriverCertificate
    Issued by: DriverCertificate
    Expires:   Sun Jan 01 07:59:59 2040
    SHA1 hash: 023C47C13373B4B4DD57A4DA46B0E5296C41FA91

The following certificate was selected:
    Issued to: DriverCertificate
    Issued by: DriverCertificate
    Expires:   Sun Jan 01 07:59:59 2040
    SHA1 hash: EA91DC78D3DCB4CCA3A42F13AC38A0AD0BE797A8

Done Adding Additional Store
Successfully signed: D:\git\github\forked\windows-drivers-rs\target\debug\package\sample_kmdf_driver.cat

Number of files successfully Signed: 1
Number of warnings: 0
Number of errors: 0
[cargo-make] INFO - Build Done in 125.14 seconds.

D:\git\github\forked\windows-drivers-rs>
```

kouzhudong, 2023-11-14:
```
D:\git\github\forked\windows-drivers-rs>cargo make --cwd ./crates/wdk-macros
[cargo-make] INFO - cargo make 0.37.2
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Project: wdk-macros
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Execute Command: "cargo" "fmt"
Warning: can't set `wrap_comments = true`, unstable features are only available in nightly channel.
Warning: can't set `format_code_in_doc_comments = true`, unstable features are only available in nightly channel.
Warning: can't set `normalize_comments = true`, unstable features are only available in nightly channel.
Warning: can't set `normalize_doc_attributes = true`, unstable features are only available in nightly channel.
Warning: can't set `format_strings = true`, unstable features are only available in nightly channel.
Warning: can't set `format_macro_matchers = true`, unstable features are only available in nightly channel.
Warning: can't set `hex_literal_case = Upper`, unstable features are only available in nightly channel.
Warning: can't set `imports_layout = HorizontalVertical`, unstable features are only available in nightly channel.
Warning: can't set `imports_granularity = Crate`, unstable features are only available in nightly channel.
Warning: can't set `group_imports = StdExternalCrate`, unstable features are only available in nightly channel.
Warning: can't set `reorder_impl_items = true`, unstable features are only available in nightly channel.
Warning: can't set `version = Two`, unstable features are only available in nightly channel.
Warning: can't set `condense_wildcard_suffixes = true`, unstable features are only available in nightly channel.
Warning: can't set `unstable_features = true`, unstable features are only available in nightly channel.
[cargo-make] INFO - Skipping Task: format-toml-conditioned-flow
[cargo-make] INFO - Execute Command: "cargo" "build" "--all-features"
   Compiling syn v2.0.32
   Compiling wdk-macros v0.1.0 (D:\git\github\forked\windows-drivers-rs\crates\wdk-macros)
error[E0554]: `#![feature]` may not be used on the stable release channel
 --> crates\wdk-macros\src\lib.rs:6:42
  |
6 | #![cfg_attr(feature = "nightly", feature(hint_must_use))]
  |                                          ^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0554`.
error: could not compile `wdk-macros` (lib) due to previous error
[cargo-make] ERROR - Error while executing command, exit code: 101
[cargo-make] WARN - Build Failed.

D:\git\github\forked\windows-drivers-rs>
```

kouzhudong, 2023-11-14:
suggestion delete 
#![cfg_attr(feature = "nightly", feature(hint_must_use))]
in crates\wdk-macros\src\lib.rs

wmmc88, 2023-11-14:
> suggestion delete #![cfg_attr(feature = "nightly", feature(hint_must_use))] in crates\wdk-macros\src\lib.rs
This is not possible because if you enable `nightly` features on the crate, we use the unstable `hint_must_use` feature.

> `cargo make --cwd ./crates/wdk-macros`

This is your mistake. The `wdk-macros` is not a driver crate. The only sample driver crate in this repository right now is `sample-kmdf-driver`, so if you're goal is just to build, package and test that, then `cargo make --cwd ./crates/sample-kmdf-driver` is all you need.

closed 2024-01-31 by wmmc88