microsoft/windows-drivers-rs #41: It looks like the issue is that the Rust code is missing a definition for the _DllMainCRTStartup symbol.

kelachim, 2023-11-14:
Error: link using `link.exe` failed: exit code: 1120
  |
  = note: "C:\\Program Files (x86)\\\Microsoft Visual Studio\\\2022\\\BuildTools\\\VC\\Tools\\\MSVC\\\14.36.32532\\\\bin\\HostX64\x64\\link. exe" "/DEF:C:\\Users\\\\kelac\\AppData\Local\\\Temp\rustciNoZXt\lib. def" "/NOLOGO" "C:\\Users\\kelac\\\AppData\Local\Temp\rustciNoZXt\symbols. o" "C:\\Users\\kelac\\OneDrive\\Desktop\\joe\\driver\\target\\debug\\deps\\driver.wdk_alloc-fab5448edb774dc7.wdk_alloc.d3d3930f6a8c5a64-cgu.0 .rcgu.o.rcgu.o" "/LIBPATH:C:\\Users\\kelac\\OneDrive\\Desktop\\joe\\\driver\\\target\\\debug\\deps" "/LIBPATH:D:\\Program Files\\Windows Kits\\\10\\Lib\\10.0.22621. 0\km\x64" "/LIBPATH:D:\\Program Files\\\Windows Kits\\\10\Lib\\\wdf\kmdf\x64\1.33" "/LIBPATH:C:\\Users\kelac\. rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "C:\\Users\\kelac\\. rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcompiler_builtins-ea81af1caf017ce1. rlib" "BufferOverflowFastFailK.lib" "ntoskrnl.lib" "hal.lib" "wmilib.lib" "WdfLdr.lib" "WdfDriverEntry.lib" "/NXCOMPATH" "/LIBPATH:C:\\Users\\\\kelac\\. rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "/OUT:C:\\Users\\kelac\\OneDrive\\Desktop\\joe\\driver\\target\\debug\\deps\\driver. dll" "/OPT:REF,NOICF" "/DLL" "/IMPLIB:C:\\Users\kelac\OneDrive\Desktop\joe\\driver\\\target\debug\deps\driver.dll. lib" "/DEBUG" "/NATVIS:C:\\Users\\\kelac\\. rustup\\\toolchains\\\nightly-x86_64-pc-windows-msvc\\\lib\\\rustlib\etc\intrinsic. natvis" "/NATVIS:C:\\Users\\\kelac\\. rustup\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\rustlib\\\etc\liballoc.natvis" "/NATVIS:C:\\Users\kelac\. rustup\\\\toolchains\\nightly-x86_64-pc-windows-msvc\\\lib\\\rustlib\\\etc\\libcore.natvis" "/NATVIS:C:\\Users\\\kelac\.rustup\\toolchains\nightly-x86_64-pc-windows-msvc\\\lib\rustlib\etc\\libstd.natvis"
  = Note: This creates the library C:\Users\kelac\OneDrive\Desktop\joe\driver\target\debug\deps\driver.dll.lib and the object C:\Users\kelac\OneDrive\Desktop\joe\driver\target\debug\deps\driver.dll.exp.
          LINK : error LNK2001: Unrecognized external symbol _DllMainCRTStartup
          C:\Users\kelac\OneDrive\Desktop\joe\driver\target\debug\deps\driver.dll : fatal error LNK1120: Number of unrecognized external types: 1

wmmc88, 2023-11-14:
Please provide more details like a minimum reproducible example, the commands that led to this error, the full build log, etc. From the limited info you've provided, it looks like you didn't set up the build script properly. `_DllMainCRTStartup` is not something that needs to be defined when the correct driver linker arguments are provided to `link.exe` (which is handled by `wdk_build` if you set up your build script based off the readme).

Redict, 2024-04-28:
Place build.rs in project root folder

kelachim, 2024-04-28:
I've solved it and forgot to close the issue, but thank you for the entry ❤️

closed 2024-04-28 by kelachim