microsoft/windows-drivers-rs #57: Drivers not working - even the samples

lurebat, 2023-12-27:
https://github.com/microsoft/Windows-rust-driver-samples/issues/7

I'm linking to the issue here in case it's not strictly relevant to the examples/is not checked there.
Please delete if it's the wrong place for this.

wmmc88, 2023-12-28:
I believe this is a dupe of #42. I suspect it is an issue with signing that dont happen in W11, and am investigating. I'm hoping to resolve this issue and bundle it with changes i'm working on in #35.

wmmc88, 2024-01-23:
This is due to using W10. If you want to use W10 right now, the patch described in https://github.com/microsoft/windows-drivers-rs/issues/42#issuecomment-1905274389 is required until https://github.com/microsoft/windows-drivers-rs/issues/82 lands.

Closing this as dupe. Tracking this issue in https://github.com/microsoft/windows-drivers-rs/issues/42

closed 2024-01-23 by wmmc88