microsoft/windows-drivers-rs #71: Proposal for Improving the Binding Generation Process for Windows Headers with bindgen

rthidfrev, 2024-01-08:
Hello,

I would like to submit this proposal for enhancing the Rust binding generation process for Windows headers. Currently, the process involves using bindgen multiple times to create separate Rust modules from different header files. This method leads to significant code duplication, as many headers share common functions and structures.

**Proposal:**
My proposal is to optimize this process by using bindgen once with a complete list of Windows headers, and then segmenting the generated binding file into multiple Rust sub-modules. This approach aims to reduce code duplication and improve the overall efficiency of the process.

To facilitate this segmentation, I suggest developing a function that analyzes the generated binding file, constructs a hashmap linking each element (function, variable, etc.) to its original header, and uses this information to appropriately divide the file into sub-modules. Additionally, this function could identify elements that bindgen was unable to convert, thus maintaining a tracking file for developers, indicating areas that require manual attention or implementation.

**Personal Context:**
I would like to note that I am a beginner in Rust and system programming. Therefore, it's possible that my proposal might not fully align with the best practices or current standards of the project. If that's the case, I apologize in advance for any inconvenience caused and would be grateful for any advice or suggestions to improve my proposal.

Thank you for your consideration and time.

Best regards

wmmc88, 2024-01-08:
Hi @rthidfrev ! While the current process does have bindgen do multiple passes, it doesnt create a seperate rust module for each header file processed. Could you point to some specific examples of the duplicates you noticed?

Right now, the bindings generated are broken down into internal modules for types, constants, and functions; so there shouldn't be any dupes rn. However, I do have some WIP work to decouple some of the larger modules in the wdk to seperate modules gated by Cargo Features, although this work is still on my fork and not in mainline yet. That WIP work is currently using a simpler approach of using bindgen's allowlisting/blocklisting on parts of the path the headers are in.

rthidfrev, 2024-01-09:
Hello,

Thank you for your response and the clarifications regarding the current process. Regarding my initial concern about duplication, let me clarify with a specific example. For instance, in the generated bindings, such as those in ntddk.rs, there are functions like KeGetCurrentIrql, originally defined in wdm.h. If bindings are generated for other headers that potentially include wdm.h, this could lead to duplication of some functions across different Rust modules. While this can be avoided with allowlists/blocklists, it represents a significant amount of work.

My initial proposal aimed to optimize this process. However, after discovering the existence of existing bindings for WDK, available at [https://microsoft.github.io/windows-docs-rs/doc/windows/Wdk/index.html](https://microsoft.github.io/windows-docs-rs/doc/windows/Wdk/index.html), generated through metadata via [https://github.com/microsoft/win32metadata](https://github.com/microsoft/win32metadata) and [https://github.com/microsoft/wdkmetadata](https://github.com/microsoft/wdkmetadata), I question the necessity of generating new bindings with bindgen. However, given the lack of metadata for WDF, we should either develop a base similar to wdkmetadata or win32metadata to obtain metadata for WDF, or continue using bindgen.

Consequently, I would like to refocus my proposal solely on the development of bindings for WDF. Creating a separate crate for WDF bindings with bindgen could be a simpler and more direct solution, avoiding the complexity of generating metadata like in projects such as wdkmetadata or win32metadata. This crate could be structured into modules, with features like Wdf_kmdf, Wdf_umdf, Wdf_DMA, etc.

To achieve this, we could use libclang to retrieve the Abstract Syntax Tree and generate specific allowlists/blocklists for each header. This would enable bindgen to generate bindings specific to the module we want to create, without worrying about duplication. For example, the wdf_io module could be based solely on elements defined in "wdfrequest.h", "wdfiotarget.h", and "wdfio.h".

I am not certain about the technical feasibility of this approach or its relevance in the current project context, and I would appreciate any feedback or suggestions on this matter.

Thank you for considering my proposal and for the time you have taken to review it.

wmmc88, 2024-01-09:
> Regarding my initial concern about duplication, let me clarify with a specific example. For instance, in the generated bindings, such as those in ntddk.rs, there are functions like KeGetCurrentIrql, [...]

Allowlist/Blocklist should not be much because lists of these headers are already required to further increase API coverage gated by [cargo features](https://doc.rust-lang.org/cargo/reference/features.html) by introducing one new generated module for each new feature. I will probably be using this approach soon to expand the api coverage of WDR.

> My initial proposal aimed to optimize this process. However, after discovering the existence of existing bindings for WDK, available at [...]

We are aware of those projects and work with the people running them, and we have different goal as outlined in https://github.com/microsoft/windows-drivers-rs/discussions/20. Our eventual goal is unification by swapping the rust apis to call bindings provided by the metadata, but there are many blockers in the metadata generation process so that is likely to be very far down the line.

> Creating a separate crate for WDF bindings with bindgen could be a simpler and more direct solution, avoiding the complexity of generating metadata like in projects such as wdkmetadata or [...]

This is similar to work that we are already planning on doing via features, as I alluded to above :)

> To achieve this, we could use libclang to retrieve the Abstract Syntax Tree and generate specific allowlists/blocklists for each header. This would enable bindgen to generate bindings specific to [...]

Based off some experimenting with bindgen's allowlist/blocklisting, it seems like clang has already resolved the header include to a full path by the time bindgen's allow/blocklisting apis are processed so its possible to allow/block based on the folders in the path. I fail to see how the added complexity of allow/blocklisting individual items in each header, rather than just allow/blocklisting headers or folders, adds any benefit. Am i missing something in your approach?