microsoft/windows-drivers-rs #97: failed to get `wdk-build` as a dependency of package

zanpocc, 2024-01-31:
I created a demo, but when running cargo make, I encountered an error. 
Below is the error message:
```
C:\Users\Administrator\Desktop\rust_driver>cargo make
[cargo-make] INFO - cargo make 0.37.8

C:\Users\Administrator\Desktop\rust_driver>cd "C:\Users\Administrator\Desktop\rust_driver"

C:\Users\Administrator\Desktop\rust_driver>pwsh.exe -Command "if ($env:CARGO_MAKE_CRATE_IS_WORKSPACE) { return };$cargoMakeURI = 'https://raw.githubusercontent.com/microsoft/windows-drivers-rs/main/rust-driver-makefile.toml';New-Item -ItemType Directory .cargo-make-loadscripts -Force;Invoke-RestMethod -Method GET -Uri $CargoMakeURI -OutFile $env:CARGO_MAKE_WORKSPACE_WORKING_DIRECTORY/.cargo-make-loadscripts/rust-driver-makefile.toml"

    Directory: C:\Users\Administrator\Desktop\rust_driver

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----            2024/2/1     5:57                .cargo-make-loadscripts

[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Project: rust_driver
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] ERROR - Error while running plugin: Source: Unknown Line: 37 - [tasks.wdk-build-init]'s script failed with exit code: 1
stdout:

stderr:
error: failed to get `wdk-build` as a dependency of package `main_f41fa1c176f19b8ce8394bac v0.1.0 (C:\Users\Administrator\AppData\Local\rust-script\projects\f41fa1c176f19b8ce8394bac)`

Caused by:
  failed to load source for dependency `wdk-build`

Caused by:
  Unable to update C:\Users\Administrator\Desktop\rust_driver\.cargo-make-loadscripts\crates\wdk-build

Caused by:
  failed to read `C:\Users\Administrator\Desktop\rust_driver\.cargo-make-loadscripts\crates\wdk-build\Cargo.toml`

Caused by:
  系统找不到指定的路径。 (os error 3)
error: Could not execute cargo

The temporary rust-script file is located at C:\Users\Administrator\Desktop\rust_driver\target/cargo-make-script/wdk-build-init/main.rs
[cargo-make] WARN - Build Failed.
```

I checked, and the wdk-build-0.1.0 dependency was successfully downloaded. 
 I also tried downloading a sample project and encountered the same error.

zanpocc, 2024-02-01:
My bad,I should use v0.1.0 branch.

closed 2024-02-01 by zanpocc

ProgrammedInsanity, 2024-02-04:
Could you share your solution in detail?

iKunCai, 2024-02-04:
[wdk-build](error: failed to get `wdk-build` as a dependency of package `main_b97ff0a4209e20c0bd6a62ad)
Same error here!
 Please share what do you mean?

zanpocc, 2024-02-13:
Sorry guys, delayed response:
I just changed the URI address in the Makefile.toml file to v0.1.0. cargoMakeURI = 'https://raw.githubusercontent.com/microsoft/windows-drivers-rs/v0.1.0/rust-driver-makefile.toml';

@iKunCai @ProgrammedInsanity

wmmc88, 2024-02-13:
With version `0.2.0` released, downloading the seperate makefile is no longer needed. Checkout the latest `load_script` snippet in the `0.2.0` readme!