microsoft/windows-drivers-rs #99: failed to load source for dependency `wdk-build`

iKunCai, 2024-02-05:
ERROR - Error while running plugin: Source: Unknown Line: 38 - [tasks.wdk-build-init]'s script failed with exit code: 1

Caused by:
  failed to load source for dependency `wdk-build`

I got same error in the closed issues.however, there is not information about how  to fix this error

lurebat, 2024-02-05:
I worked around it by changing the makefile - 

Change Makefile.toml to 
```
extend = ".cargo-make-loadscripts/rust-driver-makefile.toml"

[env]
CARGO_MAKE_EXTEND_WORKSPACE_MAKEFILE = true
```

And then in `rust-driver-makefile.toml` change every `wdk-build = ...`
to `wdk-build = {git = "https://github.com/microsoft/windows-drivers-rs.git", branch = "main"}`

iKunCai, 2024-02-05:
> I worked around it by changing the makefile -
>
> Change Makefile.toml to
>
> ```
> extend = ".cargo-make-loadscripts/rust-driver-makefile.toml"
>
> [env]
> CARGO_MAKE_EXTEND_WORKSPACE_MAKEFILE = true
> ```
>
> And [...]

thank you. i will try it.

lurebat, 2024-02-05:
I'm guessing #36 will let you edit the file, but it still needs to be fixed, so maybe don't close the issue.

iKunCai, 2024-02-05:
> I'm guessing #36 will let you edit the file, but it still needs to be fixed, so maybe don't close the issue.

Ok,Done

wmmc88, 2024-02-05:
#36 makes it so that there is not longer an extra step to get the rust-driver-makefile, since it already exists in the filesystem when u take a dependency on wdk-build. This also ensures the makefile is versioned with the build crate.

The previous loadscript(assuming u copied from the example/readme) always pulled the latest makefile from main branch. Its possible that when #35 merged, it started pulling an incompatible rust-driver-toolchain.toml. you can edit the loadscript to pull in the specific one you want. 

What version of the crates are you using, and how have you consumed them(git vs crates.io)? What does you `load_script` look like?

lurebat, 2024-02-05:
The makefile as it currently is seems to look for the crate in the wrong path.

it tries to go to .cargo-make-loadscripts/crates/wdk-build

Which makes sense for this repo (as there is a crates directory with wdk-build in the path of the project), but in repos that use it it doesn't exist.

wmmc88, 2024-02-06:
> The makefile as it currently is seems to look for the crate in the wrong path.
> 
> it tries to go to .cargo-make-loadscripts/crates/wdk-build
> 
> Which makes sense for this repo (as there is a crates directory with wdk-build in the path of the project), but in repos that use it it doesn't exist.

How does your repo use WDR? 

The `crates/wdk-build` path used [here](https://github.com/microsoft/windows-drivers-rs/blob/103ad12e34d9188b54e29fcb717fccaa746bc630/Makefile.toml#L1) is in a makefile that is meant only for use in this repo. The readme was updated in #36 (now merged) with new paths for a client's loadscript.

lurebat, 2024-02-10:
Using the newest script in the readme fixes it

closed 2024-02-12 by wmmc88