microsoft/windows-drivers-rs #107: Bindgen Clang compile error in wdk-sys - missing preprocessor defines

polloloco, 2024-02-12:
I had trouble compiling my own test driver, so I wanted to compile the sample-kmdf-driver crate to see if the problem is with my driver.

But I get the same error from that project (full output below):
```
Bindings should succeed to generate: ClangDiagnostic("E:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/ntifs.h:25:2: error: Compiler version not supported by Windows DDK...
```

I'm running Windows 10 LTSC, with the latest (10.0.22621) EWDK iso mounted to E:
It looks like the compiler doesn't set preprocessor defines such as _MSC_VER for some reason

```
C:\dev\rust-kernel\windows-drivers-rs\crates\sample-kmdf-driver>cargo make
[cargo-make] INFO - cargo make 0.37.9
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Project: sample-kmdf-driver
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: default
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
Building and packaging driver: sample-kmdf-driver...
[cargo-make] INFO - Running Task: default
[cargo-make] INFO - Running Task: build
[cargo-make] INFO - Execute Command: "cargo" "build"
   Compiling wdk-sys v0.2.0 (C:\dev\rust-kernel\windows-drivers-rs\crates\wdk-sys)
error: failed to run custom build command for `wdk-sys v0.2.0 (C:\dev\rust-kernel\windows-drivers-rs\crates\wdk-sys)`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `C:\dev\rust-kernel\windows-drivers-rs\target\debug\build\wdk-sys-83dd066b1c411ae7\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=TARGET
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS_x86_64-pc-windows-msvc
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS_x86_64_pc_windows_msvc
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS
  cargo:rerun-if-changed=src/ntddk-input.h
  cargo:rerun-if-changed=src/wdf-input.h

  --- stderr
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/ntifs.h:25:2: error: Compiler version not supported by Windows DDK
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/ntddk.h:28:2: error: Compiler version not supported by Windows DDK
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:38:2: error: Compiler version not supported by Windows DDK
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/basetsd.h:318:33: warning: cast to smaller integer type 'unsigned long' from 'const void *' [-Wvoid-pointer-to-int-cast]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/basetsd.h:327:32: warning: cast to smaller integer type 'long' from 'const void *' [-Wvoid-pointer-to-int-cast]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/basetsd.h:336:12: warning: cast to 'void *' from smaller integer type 'unsigned long' [-Wint-to-void-pointer-cast]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/ntdef.h:1390:2: error: Must define a target architecture.
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4864:19: error: call to undeclared function 'ReadAcquire8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4875:19: error: call to undeclared function 'ReadNoFence8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4886:21: error: call to undeclared function 'ReadAcquire8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4897:21: error: call to undeclared function 'ReadNoFence8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4930:5: error: call to undeclared function 'WriteRelease8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4943:5: error: call to undeclared function 'WriteNoFence8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4956:5: error: call to undeclared function 'WriteRelease8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4969:5: error: call to undeclared function 'WriteNoFence8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:4994:20: error: call to undeclared function 'ReadAcquire16'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:5005:20: error: call to undeclared function 'ReadNoFence16'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:5028:5: error: call to undeclared function 'WriteRelease16'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:5041:5: error: call to undeclared function 'WriteNoFence16'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:5066:19: error: call to undeclared function 'ReadAcquire'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:5077:19: error: call to undeclared function 'ReadNoFence'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  E:\Program Files\Windows Kits\10\Include\10.0.22621.0\km/wdm.h:5100:5: error: call to undeclared function 'WriteRelease'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
  clang diag: E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/basetsd.h:318:33: warning: cast to smaller integer type 'unsigned long' from 'const void *' [-Wvoid-pointer-to-int-cast]
  clang diag: E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/basetsd.h:327:32: warning: cast to smaller integer type 'long' from 'const void *' [-Wvoid-pointer-to-int-cast]
  clang diag: E:\Program Files\Windows Kits\10\Include\10.0.22621.0\shared/basetsd.h:336:12: warning: cast to 'void *' from smaller integer type 'unsigned long' [-Wint-to-void-pointer-cast]
  thread 'main' panicked at crates\wdk-sys\build.rs:27:14:
  Bindings should succeed to generate: ClangDiagnostic("E:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/ntifs.h:25:2: error: Compiler version not supported by Windows DDK\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/ntddk.h:28:2: error: Compiler version not supported by Windows DDK\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/wdm.h:38:2: error: Compiler version not supported by Windows DDK\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\shared/ntdef.h:1390:2: error: Must define a target architecture.\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/wdm.h:4864:19: error: call to undeclared function 'ReadAcquire8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/wdm.h:4875:19: error: call to undeclared function 'ReadNoFence8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/wdm.h:4886:21: error: call to undeclared function 'ReadAcquire8'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]\nE:\\Program Files\\Windows Kits\\10\\Include\\10.0.22621.0\\km/wdm. [... cut ...]
    27:     0x7ff687752a60 - invoke_main
                                 at d:\a01\_work\43\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    28:     0x7ff687752a60 - __scrt_common_main_seh
                                 at d:\a01\_work\43\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    29:     0x7ffde0ab7344 - BaseThreadInitThunk
    30:     0x7ffde28226b1 - RtlUserThreadStart
[cargo-make] ERROR - Error while executing command, exit code: 101
[cargo-make] WARN - Build Failed.
```

wmmc88, 2024-02-12:
What is the output of `rustup show`?

polloloco, 2024-02-12:
Yes, msvc

```
Default host: x86_64-pc-windows-msvc
rustup home:  C:\Users\User\.rustup

installed toolchains
--------------------

stable-x86_64-pc-windows-msvc (default)
nightly-x86_64-pc-windows-msvc
esp

installed targets for active toolchain
--------------------------------------

riscv32imc-unknown-none-elf
x86_64-pc-windows-msvc

active toolchain
----------------

stable-x86_64-pc-windows-msvc (default)
rustc 1.75.0 (82e1608df 2023-12-21)
```

wmmc88, 2024-02-12:
I'm unable to repro this error locally. I suspect it has to do with your installation of msvc. A couple questions:

- Is `cargo build` failing with the same error?
- Does setting `CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true` and running `cargo build` provide any additional info? 
- Are you running these commands from the ewdk developer prompt? (I'm assuming you are since I see it picking up the mounted ewdk paths)
- What version of visual studio is installed?
- Can you attach your [visual studio install configuration](https://learn.microsoft.com/en-us/visualstudio/install/import-export-installation-configurations?view=vs-2022#use-the-visual-studio-installer-ui)

polloloco, 2024-02-12:
cargo build is failing the same way. Setting the environment variable doesn't provide any additional details.

Yes I'm running from the ewdk developer prompt. I have Visual Studio 2022 Enterprise (version 17.8.6) installed.

My visual studio install configuration:
```json
{
  "version": "1.0",
  "components": [
    "Microsoft.VisualStudio.Component.Roslyn.Compiler",
    "Microsoft.Component.MSBuild",
    "Microsoft.VisualStudio.Component.SQL.LocalDB.Runtime",
    "Microsoft.VisualStudio.Component.CoreEditor",
    "Microsoft.VisualStudio.Workload.CoreEditor",
    "Microsoft.VisualStudio.Component.TextTemplating",
    "Microsoft.VisualStudio.Component.IntelliTrace.FrontEnd",
    "Microsoft.VisualStudio.Component.Debugger.JustInTime",
    "Microsoft.VisualStudio.Component.ClassDesigner",
    "Microsoft.VisualStudio.Component.GraphDocument",
    "Microsoft.VisualStudio.Component.CodeMap",
    "Microsoft.VisualStudio.Component.VC.CoreIde",
    "Microsoft.VisualStudio.Component.VC.Tools.x86.x64",
    "Microsoft.VisualStudio.Component.Graphics.Tools",
    "Microsoft.VisualStudio.Component.VC.DiagnosticTools",
    "Microsoft.VisualStudio.Component.Windows11SDK.22621",
    "Microsoft.VisualStudio.Component.VC.ATL",
    "Microsoft.VisualStudio.ComponentGroup.ArchitectureTools.Native",
    "Microsoft.VisualStudio.Component.VC.Redist.14.Latest",
    "Microsoft.VisualStudio.ComponentGroup.NativeDesktop.Core",
    "Microsoft.VisualStudio.ComponentGroup.WebToolsExtensions.CMake",
    "Microsoft.VisualStudio.Component.VC.CMake.Project",
    "Microsoft.VisualStudio.Component.VC.ASAN",
    "Microsoft.VisualStudio.Component.Vcpkg",
    "Microsoft.VisualStudio.Component.VC.Llvm.ClangToolset",
    "Microsoft.VisualStudio.Component.VC.Llvm.Clang",
    "Microsoft.VisualStudio.ComponentGroup.NativeDesktop.Llvm.Clang",
    "Microsoft.VisualStudio.Workload.NativeDesktop",
    "Microsoft.VisualStudio.Component.VC.ATL.Spectre",
    "Microsoft.VisualStudio.Component.VC.ATLMFC.Spectre",
    "Microsoft.VisualStudio.Component.VC.Runtimes.x86.x64.Spectre"
  ]
}
```

I'll try to reproduce the error in a VM tomorrow

wmmc88, 2024-02-13:
Good to hear you solved your issue!

closed 2024-02-13 by wmmc88