microsoft/windows-drivers-rs #156: Why `extern "C"`?

HoShiMin, 2024-05-11:
As we know, almost all Windows functions have `stdcall` convention, so, this is strange that all generated bindings are marked as `extern "C"`. It seems that the convention should be changed to `extern "stdcall"`, isn't it? Of course, for functions that are really "stdcall".

For example, [`ZwCreateFile`](https://github.com/microsoft/windows-drivers-rs/blob/main/crates/wdk-sys/generated_bindings/ntddk.rs#L5867).  
It has the following prototype:
```cpp
NTSYSAPI
NTSTATUS
NTAPI // <-- Here it is!
ZwCreateFile(...);
```

wmmc88, 2024-05-13:
Are you encountering any buggy behavior that is affected by this? Technically, I think the most correct solution here is that bindgen generates `extern system`:
> extern "system" -- Usually the same as extern "C", except on Win32, in which case it's "stdcall", or what you should use to link to the Windows API itself
> --From https://doc.rust-lang.org/reference/items/external-blocks.html

Functionally, there should be no difference in behavior here because [on ARM and x64 processors, `__stdcall` is accepted and ignored by the compiler](https://learn.microsoft.com/en-us/cpp/cpp/stdcall?view=msvc-170). 64-bit targets use the [x64 calling convention](https://learn.microsoft.com/en-us/cpp/build/x64-calling-convention?view=msvc-170). On Windows targets, `extern "C"` [is equivalent to `extern "win64"`](https://doc.rust-lang.org/reference/items/external-blocks.html#abi), which is Rust's moniker for Window's x64 calling convention

HoShiMin, 2024-05-13:
@wmmc88, sure, there are no problems in x64, but for the correctness it would be great to use "stdcall" or "system" as you stated above - for example, for cases of building a 32-bit driver for legacy Windows

wmmc88, 2024-05-13:
Per [our current support policy](https://github.com/microsoft/windows-drivers-rs?tab=readme-ov-file#supported-configurations):
> This project was built with support of WDM, KMDF, and UMDF drivers in mind, as well as Win32 Services. This includes support for all versions of WDF included in WDK 22H2 and newer.

Since NI WDK (22H2) does not support 32-bit drivers, we do not currently have plans to support this scenario. However, if the change to emit `extern "system"` instead of `extern "C"` is minimally invasive, then we would consider a PR.

jdoubleu, 2026-09-17:
What I found confusing was that [`DriverEntry` has to be `extern "system"`](https://github.com/microsoft/windows-drivers-rs/blob/6db419881b5e2aa62486d232904d07deb14e4b19/examples/sample-wdm-driver/src/lib.rs#L39) while [IRP MJ's must be `extern "C"`](https://github.com/microsoft/windows-drivers-rs/blob/6db419881b5e2aa62486d232904d07deb14e4b19/examples/sample-wdm-driver/src/lib.rs#L71).