microsoft/windows-drivers-rs #171: Replace `rust-condition-script-cargo_make_current_task_initial_makefile_directory-substitution` plugin with new cargo make feature

wmmc88, 2024-05-30:
With resolution of https://github.com/sagiegurari/cargo-make/issues/1081, this plugin may be no longer needed. This can be addressed in a followup

_Originally posted by @wmmc88 in https://github.com/microsoft/windows-drivers-rs/pull/143#discussion_r1621435724_

svasista-ms, 2024-08-08:
@wmmc88, I'm working on this issue. I'll raise a PR after #186 is merged.

wmmc88, 2024-08-15:
@svasista-ms On closer inspection, I believe this task is blocked by https://github.com/sagiegurari/cargo-make/pull/1132 merging. While https://github.com/sagiegurari/cargo-make/issues/1081 was resolved, it still fails to expand the args based of local testing. https://github.com/sagiegurari/cargo-make/pull/1132 should fix that.

wmmc88, 2024-08-30:
@svasista-ms you should be unblocked now

PR #208 (2024-09-05): refactor: replace directory substitution plugin with condition_script_runner_args
Closes #171 

With the recent merge of [Expand condition_script_runner_args #1132](https://github.com/sagiegurari/cargo-make/pull/1132) the `rust-condition-script-cargo_make_current_task_initial_makefile_directory-substitution` plugin is no longer needed. The `condition_script_runner_args` allows directly passing base path to the condition script runner. This PR updates the implementation by replacing the plugin with the `condition_script_runner_args`.
closed 2024-09-05 by wmmc88