microsoft/windows-drivers-rs #174: Incorrect definition for PDRIVER_UNLOAD routine

AndreiMuntea, 2024-06-08:
Hi! I noticed that in **windows_sys::Wdk::Foundation** crate

PDRIVER_UNLOAD is defined as
```
pub type PDRIVER_UNLOAD = ::core::option::Option<unsafe extern "system" fn() -> ()>;
```

but the correct C definition would be:
``` 
void DriverUnload(
  [in] _DRIVER_OBJECT *DriverObject
)
```

Note the missing DriverObject parameter.
See docs at: https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nc-wdm-driver_unload

Currently I'm working around the issue like:
```
    unsafe { driver_object.DriverUnload = Some( ::core::mem::transmute::<unsafe extern "system" fn(*mut DRIVER_OBJECT),
                                                                                                                     unsafe extern "system" fn()>(DriverUnload)); }
```

wmmc88, 2024-06-10:
Hi @AndreiMuntea . `windows_sys` is in the [`windows-rs` repo](https://github.com/microsoft/windows-rs). It is a different project that generates bindings in a different way (via winmd metadata files). 

The `windows-drivers-rs` repo is a different project that generates bindings to the WDK via `bindgen`, with the intent of allowing the same configurability that you get from the WDK.

closed 2024-06-10 by wmmc88

AndreiMuntea, 2024-06-10:
Gotcha. Thanks. Opened https://github.com/microsoft/windows-rs/issues/3087