microsoft/windows-drivers-rs #185: Missing APIs in the library

joaoviictorti, 2024-07-16:
Hi guys. 

I have a question about the library: does it contain all the APIs documented by Microsoft for Windows kernel drivers using Rust? I noticed that I didn't find some APIs in the library, like PsGetCurrentThread and Minifilters. I know that I can make the connection manually and call the function in the code, but I wondered if that was the intention of not having them. I'm thinking of making a Pull Request related to this, but I wanted to clear this doubt first.

joaoviictorti, 2024-07-18:
From what I've analyzed, the `PsGetCurrentThread` API uses the `CFORCEINLINE` macro. As the library doesn't use the `.generate_inline_functions(true)` option in the bindgen, the correct bindings are not generated for this API. The bindgen may even generate the bindings, but it ends up not putting `#[inline]`, for example.

I saw some discussions on rust-bindgen's github and saw that one person made a pull request, regarding #51 , I don't know how this one is progressing, but the question still remains regarding the file system filter apis

wmmc88, 2024-07-18:
> does it contain all the APIs documented by Microsoft for Windows kernel drivers using Rust? 

Not currently. We are planning on expanding api coverage, grouping headers into functional areas (ex. I have a branch adding all the headers related to [HID](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/_hid/). I'd consider PRs expanding coverage, but will not review them until after #186 merges. With that said, `PsGetCurrentThread` is in `ntddk.h`, which[ is already included](https://github.com/microsoft/windows-drivers-rs/blob/4ee4b8894215f43b748f5d229f7bbd88093f7986/crates/wdk-sys/src/ntddk-input.h#L5). 

> From what I've analyzed, the PsGetCurrentThread API uses the CFORCEINLINE macro. As the library doesn't use the .generate_inline_functions(true) option in the bindgen, the correct bindings are not [...]
You're right in that `PsGetCurrentThread` is not currently generated because it is an inline function. `generate_inline_functions` doesnt help here because it would have no implementation to link to. The experimental `wrap_static_fns` may work, but I haven't tested it. For more info: https://rust-lang.github.io/rust-bindgen/faq.html#why-isnt-bindgen-generating-bindings-to-inline-functions. 

#51 is largely irrelevant to this issue since that is only for WDF apis

smichaku, 2024-09-03:
Now that #186 is merged, what are the guidelines for adding new APIs?

I, for example, need the [UDECX](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/_usbref/#emulated-host-controller-driver-reference) API. This API also uses a function table `UdecxFunctions`, so like for WDF there are no direct bindings. I generalized the `call_unsafe_wdf_function_binding_impl` macro in `wdk-macros` in order to support both WDF and UDECX and updated the `wdk-sys` build to include the relevant USB and UDECX headers. I doubt this is the right way to go especially following the changes introduced by #186... Perhaps such APIs deserve a crate of their own? Or, if we still want them under `wdk-sys`, can we opt them in/out with crate features?

shujaatak, 2024-10-21:
Ah Minifilters!