microsoft/windows-drivers-rs #206: cannot find function `configure_binary_build` in crate `wdk_build` not found in `wdk_build`

Escandiuzzi, 2024-08-28:
I'm following the steps in the "Getting Started" chapter but I'm getting an error with wdk_build crate "cannot find function `configure_binary_build` in crate `wdk_build` not found in `wdk_build`" in build.rs file.

My build.rs file looks like this

`fn main() -> Result<(), wdk_build::ConfigError> {
   wdk_build::configure_binary_build()
}`

Am I missing something?

wmmc88, 2024-08-28:
If you are following the Getting Started in main branch, it corresponds with code in the main branch (currently not released to crates.io yet). Either use the git tag for `0.2.0` to read the docs for that released version of the crates, or use the latest via a [cargo git dependency](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-dependencies-from-git-repositories). If you go the git dependency route, you should pin to a `rev`.

Just an FYI that the latest main code has quite a few improvements and major changes.

Escandiuzzi, 2024-08-28:
Cool! That makes sense :)

Thank you!

closed 2024-08-28 by Escandiuzzi