microsoft/windows-drivers-rs #215: fatal error LNK1169

henceiusegentoo, 2024-09-07:
Followed the instructions in the readme and now get the following error:
```
error: linking with `link.exe` failed: exit code: 1169
  |
  = note: "E:\\Program Files\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.38.33130\\bin\\HostX64\\x64\\link.exe" "/DEF:C:\\Users\\Jimmy\\AppData\\Local\\Temp\\rustcuiKNEY\\lib.def" "/NOLOGO" "C:\\Users\\Jimmy\\AppData\\Local\\Temp\\rustcuiKNEY\\symbols.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.02h8u4mswl1ookgpj5l4kmeuh.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.37wg5pg35kstchxylvzvkwcz8.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.3r1bg36b6oy3zuj4c94w2vaz1.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.584u3kc340u9tfr7xnkxago0a.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.7brs6hr0htsgsjf94rg5k0r10.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.9pl0nzfkf1nlz55feg4y956lx.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.bj1hgsmbgknva189948yaqq5i.rcgu.o" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.egy5ollrw6r7icq9baw6qofyw.rcgu.o" "/LIBPATH:C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps" "/LIBPATH:E:\\Program Files\\Windows Kits\\10\\Lib\\10.0.26100.0\\km\\x64" "/LIBPATH:E:\\Program Files\\Windows Kits\\10\\Lib\\wdf\\kmdf\\x64\\1.33" "/LIBPATH:C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\libwdk_alloc-65c9a314840e6926.rlib" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\libwdk_sys-3fa76e0020e7441d.rlib" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\liblazy_static-6c50b0efb9151bac.rlib" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\libspin-2a11f723a40e5af1.rlib" "C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\libwdk_panic-fbe39b7b83cbc654.rlib" "C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\librustc_std_workspace_core-d08e4e7d05aff086.rlib" "C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcore-89ef43c30f9b928e.rlib" "C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcompiler_builtins-1426abde5263ac46.rlib" "BufferOverflowFastFailK.lib" "ntoskrnl.lib" "hal.lib" "wmilib.lib" "WdfLdr.lib" "WdfDriverEntry.lib" "/defaultlib:libcmt" "/NXCOMPAT" "/LIBPATH:C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "/OUT:C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.dll" "/OPT:REF,NOICF" "/DLL" "/IMPLIB:C:\\Users\\Jimmy\\Programmieren\\lsrpm-wpm\\target\\debug\\deps\\lsrpm_wpm.dll.lib" "/DEBUG" "/PDBALTPATH:%_PDB%" "/NATVIS:C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\intrinsic.natvis" "/NATVIS:C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\liballoc.natvis" "/NATVIS:C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\libcore.natvis" "/NATVIS:C:\\Users\\Jimmy\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\libstd.natvis"
  = note: libvcruntime.lib(risctrnsctrl.obj) : error LNK2005: __CxxFrameHandler3 already defined in libwdk_sys-3fa76e0020e7441d.rlib(wdk_sys-3fa76e0020e7441d.wdk_sys.46b16dd889c0d8d-cgu.0.rcgu.o)
             Creating library C:\Users\Jimmy\Programmieren\lsrpm-wpm\target\debug\deps\lsrpm_wpm.dll.lib and object C:\Users\Jimmy\Programmieren\lsrpm-wpm\target\debug\deps\lsrpm_wpm.dll.exp
          C:\Users\Jimmy\Programmieren\lsrpm-wpm\target\debug\deps\lsrpm_wpm.dll : fatal error LNK1169: one or more multiply defined symbols found
```

kittuov, 2025-05-26:
For anyone coming back here with the same error, 

This happened to me when building for UMDF.

I fixed it by:

Remove allocator, `!#[no_std]` and panic handler in `cargo.toml`

jdoubleu, 2026-09-17:
Is this still an issue with the current [`main` branch](https://github.com/microsoft/windows-drivers-rs/tree/main)?

If UMDF requires special treatment, does this need to be documented somewhere?

henceiusegentoo, 2026-09-21:
I do not know. I haven't work with this for a long while. Someone would need to check

wmmc88, 2026-09-29:
This should not be an issue anymore, and the readme has long since been updated

closed 2026-09-29 by wmmc88