microsoft/windows-drivers-rs #242: Add `WDF_STRUCTURE_SIZE` macro equivalent

wmmc88, 2024-11-07:
`WDF_STRUCTURE_SIZE` is a macro available in the WDK, but isn't available in `wdk-sys` since bindgen can't generate macros. `WDF_STRUCTURE_SIZE` is useful since many of the `_INIT` macros are also not available right now. A 1:1 implementation of `WDF_STRUCTURE_SIZE` should be made available in `wdk-sys`, although an equivalent shouldn't be needed in `wdk` given that the safe abstractions shouldn't need to do any of the c-style manual initializations.

0xflux, 2024-12-09:
I'd like to have a look at this @wmmc88. Would this live in wdk-sys/ntddk.rs?

wmmc88, 2024-12-10:
> I'd like to have a look at this [@wmmc88](https://github.com/wmmc88). Would this live in wdk-sys/ntddk.rs?

It would live in wdf.rs, however i would hold off on doing anything here yet since we're investigating whether we can look into using bindgen's new [clang_macro_fallback](https://docs.rs/bindgen/latest/bindgen/struct.Builder.html#method.clang_macro_fallback) feature

0xflux, 2024-12-11:
Ok no probs! Looking for some places to contribute so I'll keep my eyes open