microsoft/windows-drivers-rs #258: "A required privilege is not held by the client" when running cargo make

medioxor, 2025-01-03:
I followed the documentation and i've made 0 changes but i keep getting the following error when trying to build the example:

error:
```
[cargo-make] INFO - cargo make 0.37.23
[cargo-make] INFO - Execute Command: "rust-script" "target\\_cargo_make_temp\\persisted_scripts\\2438808d54148299917399bf82bbf576.rs"
Error: IoError(Os { code: 1314, kind: Uncategorized, message: "A required privilege is not held by the client." })
[cargo-make] ERROR - Unable to execute rust code.
[cargo-make] WARN - Build Failed.
```

however when i try to build in an admin prompt everything works as expected and i can successfully build the driver, curious why admin is required by wdk-build?

contents of cargo.toml:
```
[package]
name = "example"
version = "0.1.0"
edition = "2024"
build = "build.rs"

[package.metadata.wdk.driver-model]
driver-type = "WDM"

[lib]
crate-type = ["cdylib"]

[features]
default = []
nightly = ["wdk/nightly", "wdk-sys/nightly"]

[dependencies]
wdk = "0.3.0"
wdk-alloc = "0.3.0"
wdk-panic = "0.3.0"
wdk-sys = "0.3.0"

[profile.dev]
panic = "abort"
lto = true

[profile.release]
panic = "abort"
lto = true

[build-dependencies]
wdk-build = "0.3.0"
```

contents of makefile.toml:
```
extend = "target/rust-driver-makefile.toml"

[config]
load_script = '''
#!@rust
//! ```cargo
//! [dependencies]
//! wdk-build = "0.3.0"
//! ```
#![allow(unused_doc_comments)]

wdk_build::cargo_make::load_rust_driver_makefile()?
'''
```

version info:
OS Name:                       Microsoft Windows 11 Pro
OS Version:                    10.0.26100 N/A Build 26100

0xflux, 2025-01-03:
Hey, are you using the Developer Command Prompt for VS 2022 with elevated rights? Do you have the [WDK ](https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)installed / environment?

wmmc88, 2025-01-06:
> `Execute Command: "rust-script" "target\\_cargo_make_temp\\persisted_scripts\\2438808d54148299917399bf82bbf576.rs"`

You can look at the contest of this `.rs` file to see what's throwing the error. My guess is that its failing at the symlink line. Symlinking on windows generally require the user to have admin privileges. You can avoid this requirement if you enable [Developer Mode ](https://learn.microsoft.com/en-us/windows/apps/get-started/enable-your-device-for-development#developer-mode). 

This blog goes over some background about symlinks in windows, and changes allowing users to symlink without admin: https://blogs.windows.com/windowsdeveloper/2016/12/02/symlinks-windows-10/

medioxor, 2025-01-07:
thanks @wmmc88 much appreciated, i'll enable that feature its good to know

closed 2025-01-07 by medioxor