microsoft/windows-drivers-rs #299: Pull out version magic strings into configurable setting in `wdk-build`

leon-xd, 2025-02-20:
General question - have we considered pulling out constants to a separate file? There's a lot of raw str slices around the configuration setup and I usually prefer to remove magic strings. After doing some research, there's not any clippy lints I could find that enforces this so not sure if this is achievable.

_Originally posted by @leon-xd in https://github.com/microsoft/windows-drivers-rs/pull/296#discussion_r1962106894_

leon-xd, 2025-02-20:
Entire conversation in #296 is relevant

wmmc88, 2025-02-20:
I've changed the issue title from "Pull out version magic strings into configurable setting in `wdk-build`". Removing the magic strrings is more of a side-effect of implementing a configuration setting for all the different versionned headers in the WDK