microsoft/windows-drivers-rs #304: Make 'wdk' crate compatible with cfg(no_global_oom_handling)

NateD-MSFT, 2025-02-25:
The `wdk` crate today is fairly small, but has the goal of providing nice abstractions on top of `wdk-sys`.  Our samples and templates currently include `wdk` as well as `wdk-sys`.

However, the vast majority of kernel-mode drivers are expected to gracefully handle resource allocation failures.  The best way to do this in Rust is to enable `cfg(no_global_oom_handling)`, which removes all core and std functions that panic on allocation failure.  As such, I would recommend kernel-mode drivers using Rust enable this CFG; by extension, it would be nice if the `wdk` crate could compile under this restriction.

wmmc88, 2025-02-26:
What errors (and from which crates) do you see issues with right now? OTTOMH, the only issue is when alloc feature is brought into, and the way to solve that would be enhancements to our allocator (possibly via allocator_api2)

wmmc88, 2025-02-26:
Also part of closing this task should be verification. I think that just entails an addition dimension in the matrix to compile and test with the flag set. Could you some sort of todo list to this issue?

NateD-MSFT, 2025-02-26:
> What errors (and from which crates) do you see issues with right now? OTTOMH, the only issue is when alloc feature is brought into, and the way to solve that would be enhancements to our allocator (possibly via allocator_api2)

When building with cargo +nightly build, wdk as a dependency, and the following `config.toml`:

```
[build]
rustflags = ["-C", "target-feature=+crt-static", "--cfg=no_global_oom_handling"]

[unstable]
build-std = ["core", "alloc"]
```

I get the following error:

```
error[E0432]: unresolved import `alloc::ffi`
   --> F:\packages\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\wdk-0.3.0\src\print.rs:6:12
    |
6   | use alloc::ffi::CString;
    |            ^^^ could not find `ffi` in `alloc`
    |
note: found an item that was configured out
   --> F:\packages\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\alloc\src\lib.rs:237:9
    |
237 | pub mod ffi;
    |         ^^^
note: the item is gated here
   --> F:\packages\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\alloc\src\lib.rs:236:1
    |
236 | #[cfg(all(not(no_rc), not(no_sync), not(no_global_oom_handling)))]
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0425]: cannot find function `format` in module `$crate::fmt`
   --> F:\packages\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\wdk-0.3.0\src\print.rs:37:41
    |
37  |     let formatted_string = CString::new(alloc::format!("{args}"))
    |                                         ^^^^^^^^^^^^^^^^^^^^^^^^ not found in `$crate::fmt`
    |
note: found an item that was configured out
   --> F:\packages\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\alloc\src\fmt.rs:644:8
    |
644 | pub fn format(args: Arguments<'_>) -> string::String {
    |        ^^^^^^
note: the item is gated here
   --> F:\packages\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\alloc\src\fmt.rs:640:1
    |
640 | #[cfg(not(no_global_oom_handling))]
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    = note: this error originates in the macro `alloc::format` (in Nightly builds, run with -Z macro-backtrace for more info)
```

wmmc88, 2025-03-01:
https://rust-lang.zulipchat.com/#narrow/channel/219381-t-libs/topic/no_global_oom_handling.20removal/near/498629183

It looks like there is some discussion about removing `no_global_oom_handling` recently. My understanding of it seems like R4L has stopped using `no_global_oom_handling` and its alternative for guaranteeing APIs that could fail on allocation aren't called is by just not using `alloc` at all. Instead, they are rolling their own `Allocator` trait with their own collection types.

This is not to say we shouldn't do this work anyways (its fairly trivial to fix our current issues)... just bringing up discussions on `no_global_oom_handling` potentially being removed. 

cc @wedsonaf

NateD-MSFT, 2025-03-03:
Good catch! I'll read through that discussion and leran some more about how R4L is handling it.  Like I said in the first post the greater context is to try and eliminate panics (kernel crashes) from allocation failures; my plan had been to use `cfg(no_global_oom_handling)` as an early way to do this, but this might mean pulling forward more design work on the right way to handle allocations and collections (I know the `try_vec` crate has been developed/used by MS teams.)

wmmc88, 2025-03-21:
I was doing some experimenting on enabling this, and ran into https://github.com/rust-lang/rust/issues/138758. Not a blocker since its solved by using LTO, but still an issue to be aware of