microsoft/windows-drivers-rs #338: symbol linking problem

lzty, 2025-04-14:
the pub static exported variable PsInitialSystemProcess can not link properly in my environment
the linker can not find this symbol, see the following error the linker produce:
```
error: linking with `link.exe` failed: exit code: 1120
  |
  = note: "F:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Professional\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX64\\x64\\link.exe" "/DEF:C:\\Users\\admin\\AppData\\Local\\Temp\\rustc9sniWm\\lib.def" "/NOLOGO" "C:\\Users\\admin\\AppData\\Local\\Temp\\rustc9sniWm\\symbols.o" "<7 object files omitted>" "F:\\Windows Kits\\10\\Lib\\10.0.22621.0\\km\\x64\\BufferOverflowFastFailK.lib" "F:\\Windows Kits\\10\\Lib\\10.0.22621.0\\km\\x64\\ntoskrnl.lib" "F:\\Windows Kits\\10\\Lib\\10.0.22621.0\\km\\x64\\hal.lib" "F:\\Windows Kits\\10\\Lib\\10.0.22621.0\\km\\x64\\wmilib.lib" "E:\\repos\\rust\\rmtrd\\driver\\target\\debug\\deps/{librmtrd-8d7bea584602b273.rlib,libwdk-9482cf02ug\\deps/{libwdk_panic-1d96f3f6af4300f7.rlib}" "D:\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib/{librustc_std_workspace_core-ff8db640d177ed2a.rlib,libcore-745ff350a54e4299.rlib,libcompiler_builtins-b2f39c5f2779068f.rlib}" "/defaultlib:libcmt" "/NXCOMPAT" "/LIBPATH:F:\\Windows Kits\\10\\Lib\\10.0.22621.0\\km\\x64" "/OUT:E:\\repos\\rust\\rmtrd\\driver\\target\\debug\\deps\\driver.dll" "/OPT:REF,NOICF" "/DLL" "/IMPLIB:E:\\repos\\rust\\rmtrd\\driver\\target\\debug\\deps\\driver.dll.lib" "/DEBUG" "/PDBALTPATH:%_PDB%" "/NATVIS:D:\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\intrinsic.natvis" "/NATVIS:D:\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\liballoc.natvis" "/NATVIS:D:\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\libcore.natvis" "/NATVIS:D:\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\libstd.natvis" "/DRIVER" "/NODEFAULTLIB" "/SUBSYSTEM:NATIVE" "/KERNEL" "/ENTRY:DriverEntry" "/NXCOMPAT" "/DYNAMICBASE" "/MAP" "/MAPINFO:EXPORTS" "/OPT:REF,ICF" "/INTEGRITYCHECK" "/MANIFEST:NO"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
             Creating library E:\repos\rust\rmtrd\driver\target\debug\deps\driver.dll.lib and object E:\repos\rust\rmtrd\driver\target\debug\deps\driver.dll.exp␍
          driver.5zi1ddmti52blnbkc4als71vk.rcgu.o : error LNK2019: unresolved external symbol PsInitialSystemProcess referenced in function DriverEntry␍
          E:\repos\rust\rmtrd\driver\target\debug\deps\driver.dll : fatal error LNK1120: 1 unresolved externals␍
```
the code for testing as follows:
```
unsafe {
    if PsInitialSystemProcess.is_null() {
        println!("Oh my God");
    }
}
```
compilation environment:
```
rustc 1.85.0 (4d91de4e4 2025-02-17)
rustc 1.87.0-nightly (4d30011f6 2025-03-15)
wdk-sys 0.3.0 (the same as other wdk-xxx depends)
OS: windows 10 22H2 (build 19045.5608)
WDK: 10.0.22621.2428
```
i also tried the nightly rust but the problem remains
i use the following alternative approach for months to get around this problem
```
static PsInitialProcess: AtomicPtr<_KPROCESS> = AtomicPtr::new(ptr::null_mut());

// this function returns PsInitialSystemProcess
// the symbol PsInitialProcess can not be linked properly in this version of wdk-sys, don't know why
pub fn get_initial_system_process() -> PEPROCESS {
    unsafe {
        let mut process = PsInitialProcess.load(Ordering::Acquire);

        if !process.is_null() {
            return process;
        }

        let status = PsLookupProcessByProcessId(4 as _, &mut process);

        if nt_success(status) {
            let _ = PsInitialProcess.compare_exchange(
                ptr::null_mut(),
                process,
                Ordering::SeqCst,
                Ordering::SeqCst,
            );

            // is safe to dereference here, since the PsInitialProcess is never destoryed before system shutdown
            ObfDereferenceObject(process.cast());

            return process;
        }
    }

    ptr::null_mut()
}
```
but it looks weird, right ? ^_^

0xflux, 2025-05-02:
I'm getting the same issue; did you come up with any solution or workaround @lzty?

lzty, 2025-05-04:
> I'm getting the same issue; did you come up with any solution or workaround [@lzty](https://github.com/lzty)?

i've pasted the code in the issue that i used to get around this problem @0xflux 

> static PsInitialProcess: AtomicPtr<_KPROCESS> = AtomicPtr::new(ptr::null_mut());
>
> // this function returns PsInitialSystemProcess
> // the symbol PsInitialProcess can not be linked properly in this [...]

although the code seems ugly, but works in my case, please let me know if u have a better one

0xflux, 2025-05-04:
I ended up finding the symbol offset at runtime with some pointer magic; more convoluted than your approach but works for now until this is fixed :)