microsoft/windows-drivers-rs #351: [cargo-wdk] Ensure cargo-wdk and the build script uses the same version of wdk-build and other deps

krishnakumar4a4, 2025-05-12:
Both cargo-wdk and the build script in the user's driver need to get cargo metadata from the project. The metadata needs to match in both cases. However, what if you only upgrade the cargo-wdk without changing your project then the two can diverge.

Make sure the divergence doesn't occur by making cargo-wdk backward compatible with older projects.

igs-raiderler, 2025-05-13:
Great implementation of this feature. I particularly like the approach to error handling.

gurry, 2026-10-06:
Notes from a meeting where we discussed scenarios for metadata version evolution:
```
Cargo.toml      wdk-metadata in project      wdk-metadata in cargo-wdk    Outcome
----------------------------------------------------------------------------------------------
Field added (0.1 -> 0.2)
0.1              0.1                         0.2 (adds a new field)       Pass
0.1              0.2                         0.2                          Fail - ask to upgrade metadata section / or go with a default
0.1              0.2                         0.1                          Fail - ask to upgrade cargo-wdk
0.2              0.2                         0.1                          Fail - ask to upgrade cargo-wdk
0.2              0.1                         0.2                          Fail - ask to upgrade project wdk-metadata
0.2              0.2                         0.2                          Pass

Field removed (0.2 -> 0.3)
0.2              0.3                         0.3                          Fail - error says unknown field
0.3              0.2                         0.3                          Fail - error says missing field
0.3              0.3                         0.2                          Fail - error says missing field
0.3              0.3                         0.3                          Pass

Field moved across nesting levels (0.3 -> 0.4)
0.3              0.4                         0.4                          Fail - error says unknown/missing field
0.4              0.3                         0.4                          Fail - error says unknown/missing field
0.4              0.4                         0.3                          Fail - error says unknown/missing field
0.4              0.4                         0.4                          Pass

wdk-metdata in cargo-wdk >= wdk-metdata in project >= metdata section
```