microsoft/windows-drivers-rs #355: HID hidport.h headers missing for UMDF build

elupus, 2025-05-14:
According to the examples for HID miniport drivers. It should be possible to implement HID miniport drivers with UMDF builds.

Example here: https://github.com/microsoft/Windows-driver-samples/tree/main/hid/vhidmini2

This requires types from hidport.h that has been marked as only available in kernel mode drivers here: https://github.com/microsoft/windows-drivers-rs/blob/b82b4278f7085cfb0165d7cc6ffd6e0ccaad41ff/crates/wdk-build/src/lib.rs#L753

That limitation should likely be lifted for generation of bindgen headers.

wmmc88, 2025-05-14:
Interesting. `hidport.h`'s header comments specify its KM only, but the sample umdf hid driver you link does use it. Looks like they specify an extra include path (i.e. `DDK_INC_PATH`) to make it work. I think it seems reasonable to add this

wmmc88, 2025-05-14:
@elupus would you like to submit a pr with this change? 

I think we should support this header in UMDF without adding this specific `km` folder to the um [include paths](https://github.com/microsoft/windows-drivers-rs/blob/b82b4278f7085cfb0165d7cc6ffd6e0ccaad41ff/crates/wdk-build/src/lib.rs#L379). This means that in `hid_headers()`, for umdf it'd include something like `xyz/km/hidport.h` but where `xyz` is something like `C:/Program Files (x86)/Windows Kits/10/Include/10.0.26100.0/` derived from `Config`