microsoft/windows-drivers-rs #357: [cargo-wdk] Clarify terminology in Readme

krishnakumar4a4, 2025-05-21:
Rust has its own meaning to the terms package, crate & workspace. WDR has domain terms like package & project used in different contexts. Package has become an overloaded term that is used to indicate final driver package as well as used in the context of packaging flow. 
So, it's best to clarify in the Readme explicitly, the terms we plan to use and their meaning in the WDR context.

gurry, 2025-05-21:
We could add a `Terminology` section in the README explaining how we use these terms and then ensure they are used consistently across the code base.

wmmc88, 2025-05-30:
From our discussion yesterday, I think our current thinking is to continue using `Package` for Cargo packages and use `Driver Package` for the collection of files that `cargo-wdk` spits out for Windows to install

gurry, 2025-06-19:
May I propose using the term "crate" for cargo packages? Some reasons to support that are:

- "crate" is already used interchangeably with "package" in the Rust community. It is in fact a more common than "package" in my experience. The main Rust registry for example is called crates.io not packages.io.
- We too use that term in our repo .e.g. the folder containing all the cargo packages is called "crates" not "packages".
- Currently we don't have any cargo package containing multiple crates. We might have one in the future but even then I don't think using the term "crate" will cause much confusion.

Once we go with "crate" then "package" becomes available for Windows driver packages and it will be super clear what these terms mean.