microsoft/windows-drivers-rs #365: If CWD is anywhere in a crate that is in a workspace, it should build and package the entire workspace

svasista-ms, 2025-05-31:
as discussed, this is incorrect. if CWD is anywhere in a crate that is in a workspace, it should build and package the entire workspace. Please open a follow up issue

_Originally posted by @wmmc88 in https://github.com/microsoft/windows-drivers-rs/pull/306#discussion_r2116847801_

wmmc88, 2025-06-02:
Looking into this a little bit more, it seems that my statement was only correct if `--workspace` was passed in?

Either way, i think someone needs to explicitly look into behaviors, and ensure it matches normal cargo workspace behaviors

gurry, 2025-07-07:
I checked the behaviour of `cargo build`. It is as follows:

- When you are at the workspace root it builds the entire workspace regardless of whether `--workspace` is specified or not.
- When you are inside a crate member it builds the entire workspace if `--workspace` is specified and builds only that member otherwise.

The current behaviour of `cargo-wdk` matches this except that we do not have any `--workspace` option and therefore there is no way to build the entire workspace from inside a crate. We could add this option later. It does not seem urgent to me.

krishnakumar4a4, 2025-08-01:
Created issue for the --workspace flag #433

closed 2025-08-01 by krishnakumar4a4