microsoft/windows-drivers-rs #435: Fail if target overriden by config.toml or env variable

krishnakumar4a4, 2025-08-01:
Problem: If there is a config.toml with target override, then cargo build honours that but our driver package wouldn't. 

This is a mismatch that needs to be prevented by failing early for now.

gurry, 2025-10-22:
A fix for this will also fix https://github.com/microsoft/windows-drivers-rs/issues/89

gurry, 2025-12-11:
> > A fix for this will also fix [#89](https://github.com/microsoft/windows-drivers-rs/issues/89)
> 
> Solving this issue doesn't necessarily close [#89](https://github.com/microsoft/windows-drivers-rs/issues/89). [#89](https://github.com/microsoft/windows-drivers-rs/issues/89) would also require all the packaging steps to support multiple targets at once

If the requirement of #89 to be able to build a single workspace containing projects with varying targets, then that will be possible after https://github.com/microsoft/windows-drivers-rs/pull/494 because the packaging steps will also use the target that's specified on `config.tom`

gurry, 2026-01-07:
> The original intention of [#89](https://github.com/microsoft/windows-drivers-rs/issues/89) was to support multiple targets simultaneously for a single package. Cargo natively supports that for - - target and in config toml. So it still looks like multiple targets at once in the same way cargo build supports it is not fulfilled?

You are right. #494 does not enable multiple targets for a single package. It only enables having a workspace containing packages with different targets. Sorry for misunderstanding the intention of #89. 

That said it is still okay to close this issue (#435) so I'll let #494 do that. #89 will remain open.