microsoft/windows-drivers-rs #482: Validate parameters supplied to the cargo-wdk build command

svasista-ms, 2025-08-26:
track this in an issue assigned to somebody so that we don't lose track of these

_Originally posted by @wmmc88 in https://github.com/microsoft/windows-drivers-rs/pull/462#discussion_r2299774350_

wmmc88, 2025-08-26:
@svasista-ms when you create these issues from PR comments, please make sure there is sufficient context in them

svasista-ms, 2025-08-27:
The build subcommand currently constructs and runs BuildAction with minimal upfront checks. Many user errors aren’t surfaced until after WDK detection, PATH setup, or partial per‑package iteration. Early validation will produce faster, clearer failures and simplify BuildAction. 

Currently CLI parses flags (profile, target-arch, verify-signature, sample) but does not:
- Check driver intent flags (--verify-signature, --sample) against actual driver packages.
- Check that selected target arch is installed. (Nice to have)
- Check the mode of the workspace earlier, i.e. Standalone, Workspace or Emulated. (Happens inside `BuildAction::run`)

We could add a preflight for `build` in the CLI layer which can:
1. Absolutize the working directory
2. Determine workspace mode: Single/Workspace or Emulated 
3. Load Cargo Metadata Once (for Single/Workspace Mode) and use it in BuildAction. If there is no Metadata we could check for emulated workspace scenario.
4. From the workspace we could identify the members that are driver projects (i.e `metadata.wdk` is present and crate type is `cdylib`). If there are no driver members, the `--verify` and `--sample` flag do not make sense. It can be ignored.
5. Optionally we could also probe `rustup target list --installed` to check if target is installed and fail early. Otherwise, we could also let `cargo build` fail with the same reason. 
6. We could also reorder some of `BuildAction::run` method to perform `WDK Build #` detection and `PATH` setup only after confirming at least one driver package will be built.

Although these changes aren't just "validating parameters" this would clarify the ownership of each layer:
- CLI validates input and the working environment (standalone/workspace/emulated)
- Action layer operates on validated inputs

Fetching cargo metadata would happen once prebuild and will be used throughout the `build` command.